makermoekoe/Hotplate-Soldering-Iron #3: Several problems , some questions

LordFennec, 2020-01-28:
Hello everyone. Congratulations on the project. I tried to make it but I ran into several problems.

Software: Arduino IDE 1.8.10 or Arduino IDE 1.6.13
Board: wemos D1 R1

1- with the max6675 library I have problems of this type:
F: \ Arduino-1.8.10 \ libraries \ MAX6675-library-master \ max6675.cpp: 9: 24: fatal error: util / delay.h: No such file or directory

  #include <util / delay.h>

Once the max6675.cpp library has been edited by deleting this string, I can compile it correctly, but it seems to me that it can correctly read the temperature

2- When I set a temperature with the potentiometer and press the button, nothing happens.

3- What kind of thermocouple did you use?

Where am I wrong? Tnks!!

[image]

LordFennec, 2020-02-04:
Is there anyone who managed to make it work ??

LordFennec, 2020-02-04:
is there anyone who managed to make it work ??

exiledyorkie, 2020-02-04:
Hi, I didn’t have problems with the lib, nor have to edit it but I switched out the Wemos for a Nano. I got it all working but I had a problem with the screens switching and unable to press the button. Turned out to be the power supply in the end. The thermo used is a K type. If nothing happens when the button is pressed, I’d assume there’s a connection issue and/or an issue with the Thermocouple, as the number looks rather large

LordFennec, 2020-02-04:
Hi and Tnks for your reply.
But how did you do to upload on a wemos using a Nano on the IDE?

exiledyorkie, 2020-02-05:
I didn’t. I uploaded onto a Nano via the Arduino IDE. I just adjusted the code slightly.

LordFennec, 2020-02-05:
WOuld you be kind enough to share the code or the modification you made? TNKS

makermoekoe, 2020-02-05:
Hey, sorry for the late response..
Can you tell me what PSU you are using? There are people out there which have used very noisy PSUs directly next to the Wemos D1 mini and the Wemos doesn't like it haha

LordFennec, 2020-02-05:
There is something that escapes me.
I have to program a Wemos D1 as if it were an Arduino Nano, okay?
In the Arduino IDE, select Arduino Nano as a board, do I choose the COM port. And the type of programmer?
Do I connect the wemos directly to the USB port or do I have to use the PINs on board?

makermoekoe, 2020-02-05:
Ooh no, there is something wrong in your mind... You have to choose Wemos D1 Mini board. When you couldn't find it in your board list, you have to [add ESP8266 via the boardmanager](https://github.com/esp8266/Arduino).

LordFennec, 2020-02-05:
No. then we didn't understand each other from the beginning.
I already have the cards installed in the card manager. Including esp8266 and wemos.
If I choose the wemos card I have compilation errors.
If I choose Arduino Nano as the board, No. But I can't load it on the wemos board
[image]

LordFennec, 2020-02-05:
Board manager and library all update

makermoekoe, 2020-02-05:
Try choosing the right board :D It is 
LOLIN(WEMOS) D1 R2 & mini

LordFennec, 2020-02-05:
I tried with all the wemos versions available and the generic ESP8266. Always the usual error message on the MAX6675 library referred to

.... Arduino \ libraries \ MAX6675 \ max6675.cpp: 30: 14: error: '_delay_ms' was not declared in this scope

    _delay_ms (1);

LordFennec, 2020-02-05:
Can you show me your configuration?

exiledyorkie, 2020-02-05:
Hi. What I meant is, I switch to a Nano board. Not tried to upload Nano code to a D1 Mini. You have to set the board manager to the correct board. 
Sorry, I have not got access to my code or images currently, but I followed the schematics exactly as they’re displayed in the tutorial.

exiledyorkie, 2020-02-05:
Which MAX6675 lib are you using?

makermoekoe, 2020-02-05:
[image]
And yes, make sure you download the correct MAX6675 lib. It is the Adafruit one which @exiledyorkie has mentioned above.

LordFennec, 2020-02-05:
I tried them all ... but nothing to do ... I can't compile it correctly.
Is there the possibility of having the firmware already compiled (.bin)? So I avoid using the arduino IDE?

exiledyorkie, 2020-02-05:
Tried them all, are you referring to the libs or board settings?

LordFennec, 2020-02-05:
Yes...

exiledyorkie, 2020-02-05:
Sorry I’m struggling to help here. Yes doesn’t answer the question to what I asked.

LordFennec, 2020-02-05:
As I explained above ... I tried the libraries indicated and with all the cards you suggested

LordFennec, 2020-02-05:
I summarize:

1- Uninstalled all the arduino environment.

2- Reinstalled version 1.8.11

3- ESP8266 boards added

4- Added adafruit ssd1306, GFX and max6675 libraries (latest versions)

5- Compiling the code on the Wemos D1 R1 board

_Error on <util / delay.h> (library max6675)_

6- Modified max6675.cpp file as follows:

_#elif defined (ESP8266)
   #include <pgmspace.h>
#define _delay_ms (ms) delayMicroseconds ((ms) * 1000)
#endif
// # include <util / delay.h>_

7- Code recompilation OK. No error

8- Temperature reading always at 0 ° (probable incorrect thermocouple). Same error if thermocouple disconnected.

Last fault found: temperature value writing moved too low on the display

relax81, 2020-02-06:
For the issue with your text being too low:

I guess it's handling your oled as a 128x32 and not 128x64

Try this
```
`#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels`
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);
```

LordFennec, 2020-02-06:
Rightly .... I forgot to set it. now it's OK!!
I just need to check with another thermocouple k that should come to me tomorrow. We hope to solve everything

relax81, 2020-02-07:
One issue i was running into (not sure where it comes from) was that my temperature wasn't updated. I had to add a little delay to get it to work. Maybe i've too much noise or a bad max6675 (still usign perf board until the pcbs arrive)

```
#define SENSOR_SAMPLING_TIME 300
unsigned long nextRead;

void setup() {
 nextRead = millis();
}

void loop() {
  if (millis() > nextRead)
    {
      nextRead += SENSOR_SAMPLING_TIME;
      temp_now = thermocouple.readCelsius();
    }
}
```

Some other projects switched to the newer MAX31855 - maybe i'll try that one on a second board.

LordFennec, 2020-02-08:
Oohhh finally !!! I came to the head of this intrigue.

- Pre-Test: K thermocouple probe replaced but temperature result always at 0

- Final test: Rewired the whole circuit on breadboard. Test OK

- Conclusion: Integrated max6675 fault (tracks tested ok)

Replacing it will solve the problem (I hope).

However, all my changes to the max6675.cpp library remain valid, with the original adafruit library, it does not work.
I'm really happy. Considering that I understand absolutely nothing about programming. I did some experience :-)

[image]

LordFennec, 2020-02-09:
All Oookk!! I finally did it!

[image]

closed 2020-02-09 by LordFennec